esp8266/arduino-esp8266fs-plugin #1: Error message - SPIFFS Error: esptool not found!

rogerclarkmelbourne, 2015-11-13:
Hi,

I've tried to use this but I get this error message

SPIFFS Error: esptool not found!

Setup is Windows 1.6.5 
cloned repo from github
Repo is installed and works fine, and uploads sketches

However from what I can tell, by looking in the source, this means that the plugin can't find 

File(PreferencesData.get("runtime.tools.esptool.path"), esptoolCmd);

I can't see anything like runtime.tools.esptool in the preferences.txt file

I'm not sure if I missed something in the installation instructions.

is Arduino\tools\ESP8266FS\tool supposed to also contain exptool.exe etc rather than just the JAR file?

igrr, 2015-11-13:
@h4rm0n1c Has reported success on Windows using this latest version, so it's interesting that this fails for you.

If you look at the [source code](https://github.com/esp8266/arduino-esp8266fs-plugin/blob/master/src/ESP8266FS.java#L174-L180), it checks 3 possible esptool locations:
- `{runtime.platform.path}/tools/esptool.exe` (this is where esptool used to be placed in the old IDE builds)
- `{runtime.platform.path}/tools/esptool/esptool.exe` (this is where esptool is placed now by `get.py` script)
- `{runtime.tools.esptool.path}/esptool.exe` (this where esptool is placed when you use Boards Manager package installation

Same happens for `mkspiffs` tool.

Could you please check that you have esptool.exe in `<arduino_dir>\hardware\esp8266com\esp8266\tools\esptool\esptool.exe`?

rogerclarkmelbourne, 2015-11-13:
Hi Ivan

I looked in the code but I was not sure the absolute paths of things like runtime.playform.path

I have the esptool.exe in

Arduino\hardware\esp8266com\esp8266\tools\esptool

I tried copying it to where the plugin JAR is but that didnt work either

I will try installing on another machine using the Boards manager and see if it works

I guess I'm probably going to need to rebiuild the JAR and add some more debug. 
I have java jdk1.8.0_45 on my machine, and also Make, so I should be able to rebuild it.

But it looks like I need some other Jars e.g. Arduino Core, etc, which I will need to find where to download them from.

igrr, 2015-11-13:
I haven't added build instructions to Readme yet, sorry. Here's how it goes:
- get a copy of Arduino 1.6.6 source
- Build Arduino: `cd build; ant build`
- Place a copy of this plugin into tools folder: `cd shared/tools; git clone https://github.com/esp8266/arduino-esp8266fs-plugin.git ESP8266FS; cd ESP8266FS`
- Build this plugin: `./make.sh`
You can take a look at .travis.yml :)

rogerclarkmelbourne, 2015-11-13:
OK.
I have Cygwin on one of my machines, so I may be able to use that.

I can dual boot into Linux and I have an OSX machine as well, but none of them have JDK on them.

In the mean time, I'm just running mkspiffs from the command line ;-)

Thanks again

h4rm0n1c, 2015-11-13:
Are you copying hardware and tools into the actual Arduino program folder?
Not the one in your user documents folder.

I ask because I've played with putting everything in the user specific Arduino folder and it made the error message come back while still appearing to otherwise be installed properly in the IDE.

Without more info, that's all I've got.

rogerclarkmelbourne, 2015-11-13:
I have several third party cores installed into My documents\Arduino\hardware including the esp8266 core.

And i added a tools folder to My documents\Arduino\ and unzipped the Jar file folder structure into the tools folder.

The new tool them appeared in the IDE after I restarted the IDE.

But it looks like wherever the tool is looking, for esptool etc its not in the My Documents\Arduino\hardware\esp8266 tree.

I think I need to rebuild it and get it to print out the addresses of the paths its looked in

rogerclarkmelbourne, 2015-11-13:
Ivan

Why do I need to build the IDE ?  Don't I just need the JARs referenced in the make file ?

igrr, 2015-11-13:
Um, yes, you can just download a built copy of Arduino and extract jars from there.
I was describing my development environment where i need to add some debugging to Arduino code occasionally, but for your purposes you can use pre-built jars.

rogerclarkmelbourne, 2015-11-13:
(lol) too late.

I've rebuilt it now ;-)

I'll have a go at building the plugin

Thanks

rogerclarkmelbourne, 2015-11-13:
umm

Can't get the actual plugin to compile despite finding where those jars are located

As far as I can tell.

What make.sh does is look for

```
pde_path=`find ../../../ -name pde.jar`
core_path=`find ../../../ -name arduino-core.jar`
lib_path=`find ../../../ -name commons-codec-1.7.jar`
```

Then uses those dirs as the classpath, in 

javac -target 1.8 -cp "$pde_path:$core_path:$lib_path" -d bin src/ESP8266FS.java

I manually looked for those jars on Windows and they are in 

C:\Program Files (x86)\Arduino\lib

So as far as I can tell the command to compile should be

javac -target 1.8 -cp "C:\Program Files (x86)\Arduino\lib" ESP8266FS.java 

However it doesn't seem to find any of the classes that I have referenced using -cp so I have a load of undefined for things like

ESP8266FS.java:37: error: package processing.app does not exist
import processing.app.PreferencesData;

:-(

me-no-dev, 2015-11-13:
so you go to where your arduino app is, then into it's tools folder. That folder will have one TXT file and a Mangler example.
open terminal into that folder, then:
```
git clone https://github.com/esp8266/arduino-esp8266fs-plugin esp8266fs
cd esp8266fs/
./make.sh 
```

igrr, 2015-11-13:
> On Linux, pde_path ends with pde.jar, i.e. it's a file path, not a folder
> path (lib).

Also I think the difference in our setups might be that you put the ESP8266
core into documents\Arduino\hardware, while I put it into
<ArduinoAppFolder>/hardware.

rogerclarkmelbourne, 2015-11-13:
OK.

Well, part of the problem transpired to be the Windows syntax for the java compiler

it uses semicolons to separate paths !

So if I copy the 3 jars that it needs into the source folder and do

```
javac -target 1.8 ESP8266FS.java -cp "pde.jar;arduino-core.jar;commons-codec-1.7.jar"
```

It no longer has the issues of not finding things like processing

However I now get this error

```
ESP8266FS.java:111: error: method createTempFolder in class FileUtils cannot be applied to given types;
      File buildFolder = FileUtils.createTempFolder("build", DigestUtils.md5Hex(s.getMainFilePath()) + ".spiffs");
                                  ^
  required: no arguments
  found: String,String
  reason: actual and formal argument lists differ in length
1 error
```

And looking in the code I see 

```
      // Arduino 1.6.5 doesn't have FileUtils.createTempFolder
      // String buildPath = BaseNoGui.getBuildFolder().getAbsolutePath();
```

So my guess is the reason that its not working is probably that it can't find the folders at all in 1.6.5

i.e its not even compiling using the 1.6.5 pde.jar

I can't upgrade to 1.6.6 because when I tried, it broke so many things I had to go back to 1.6.5

I may eventually upgrade to 1.6.6 but not until they have fixed a lot of the issue that everyone is reporting.

In the mean time, I guess I need to find a replacement for getBuildFolderPath

rogerclarkmelbourne, 2015-11-13:
Just a quick update

It looks like this is an issue caused because the jar is compiled against 1.6.6 and I'm running 1.6.5

I have modified the source to use the old reflection method to call getBuildFolder and it now seems to work OK

(But the file is 8k instead of 6k)

igrr, 2015-11-14:
I don't see how the issue with getBuildFolderPath could cause "esptool not found" error.
Also I did test this plugin with 1.6.5, and there is a workaround for 1.6.5 in the code (it uses reflection). I was compiling against 1.6.6.

rogerclarkmelbourne, 2015-11-17:
Hi Ivan

I think my compile issues because I'm using the 1.6.5 JAR files is not the same reason the pre-compiled plugin does not work.

I think there must be another difference e.g. the path returned by one of the other API calls.

Sorry, but at the moment I dont have time to rebuild using the 1.6.6 JAR files and add debugging information.
But now that I recompiled using the 1.6.5 JARs it works fine.

One small enhancement I think may be good, is to get the status of the serial console window, if possible, and close it prior to upload, because at the moment the upload fails if it is open, because esptool cant open the port

GarveyZuo, 2019-02-25:
I meet the same problem

GarveyZuo, 2019-02-25:
arduinos version is 1.8.8 and there's no esptool.exe in the /tool/esptool/  folder . the new version of esp8266 tools only have the esptool.py file.

ruigmoreno, 2019-05-15:
I had the same problem.

I solved it by replacing the file esp8266.jar with the new file esp8266.jar made available on the site [here](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#uploading-files-to-file-system). After that, I performed the steps in the Arduino IDE (Tools> ESP8266 Sketch Data Upload)
and it worked!

My setups:
- Ubuntu 18.04.2 LTS Gnome 3.28.2
- Arduino 1.8.9
- Board Manager:[ esp8266 by ESP8266 Community version 2.5.1](https://github.com/esp8266/Arduino#installing-with-boards-manager)
- [esptool v.2.6](https://github.com/espressif/esptool)
- [esp8266 Arduino core - ESP8266FS-0.4.0.zip](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#uploading-files-to-file-system)

chimadan, 2019-12-04:
я так же сталкиваюсь с этим