esp8266/arduino-esp8266fs-plugin #2: Subdirectories not included?

bachi76, 2015-11-15:
It seems that the upload tool does only upload files in the data folder, but not any subdirectories. Would be great if subdirs would also work. Or if that's a prob, suggest to clearly mention this limitation.

Btw is there a way / example of doing this task manually (I'm using the Eclipse Arduino IDE so need to work with two IDEs for uploads).

igrr, 2015-11-15:
@bachi76 this plugin is a thin wrapper around two tools: [mkspiffs](https://github.com/igrr/mkspiffs) and [esptool-ck](https://github.com/igrr/esptool-ck). mkspiffs is called to pack files into SPIFFS image, and esptool-ck is used to upload this image. You can certainly call these two tools from a script/makefile. Please check the readme for each tool and ping me if anything is not clear.
You will also need SPIFFS image layout, see boards.txt for parameters ([these ones](https://github.com/esp8266/Arduino/blob/master/boards.txt#L125-L128) are for 4M(3M SPIFFS) option).

Regarding recursive packing: it was added to mkspiffs by @pgollor in [2ecc11f](https://github.com/igrr/mkspiffs/commit/2ecc11f76955f512ef200b59929f7a2e4acde052), that happened between releases 0.1.1 and 0.1.2. If you got your copy of esp8266 core through boards manager package, then you've got 0.1.1, which is missing this feature. Latest git version of the core has 0.1.2.

I'm closing this issue because recursive upload feature doesn't depend on *this* tool. Feel free to open an issue against mkspiffs if anything's wrong with recursive upload in 0.1.2.

closed 2015-11-15 by igrr

bachi76, 2015-11-16:
Thanks a lot @igrr , will try with 0.1.2. (and the two underlying tools)