esp8266/arduino-esp8266fs-plugin #3: automatic gz files like css and js / or pre-create-hook

onkelfunny, 2015-11-19:
to save some space in the spiffs it would be nice if the plugin gz files like css
OR
define a pre-create-hook (bash script) to do thinks like that... it would be great if the script can change the data directory.

in my case i have 5 steps:
1. copy all files from data to /tmp/data....
2. stitch all css and js files in one css and js file
3. gz the generated files
4. remove all unused and stitched css and js files
5. build the spiffs from /tmp/data..... and upload

andig, 2016-04-06:
+1 for a pre-create hook.

igrr, 2016-04-06:
Where will this hook script will be located? In the same `data` directory?

andig, 2016-04-06:
> Where will this hook script will be located? In the same data directory?

That would make it part of the spiffs creation- probably not so good. Could we e.g. put it one level higher into build_hooks or something and can that folder be excluded from arduino ide trying to compile it?