esp8266/arduino-esp8266fs-plugin #4: Add support OTA uploading.

EUA, 2015-11-21:
This plugin tries to program thru serial port even if OTA selected.
I can upload using espota.py -i adress -p port -s -f spiffs.bin
It cannot be hard to support this.

me-no-dev, 2015-11-21:
It actually is. The data containing the network information for the selected network device does not get to the plugin, therefore we do not know port, password and other information that we need to execute the proper command.

EUA, 2015-11-21:
At least we know the IP address of the selected device as "serialPort" string.
Couldn't we use it with no password + default 8266 port configurations at least for now?

andig, 2016-01-04:
+1 for getting SPIFFS via OTA

andig, 2016-01-12:
@igrr would it suffice to modify espota.py? I'd be happy to attempt a PR if that's the direction to go.

igrr, 2016-01-12:
The plugin doesn't get the data (IP, port, password) from IDE, so it can not pass it to espota.py. If you can hack a solution using reflection or some other tricks, you're welcome. But as far as I can tell there is no "clean" way to get this data from plugin.

andig, 2016-01-12:
Based on 
> At least we know the IP address of the selected device as "serialPort" string.

I was just gonna guess if serialPort might be an IP. Could you highlight where the built espfs.jar need be placed for the ide to pick it up?

me-no-dev, 2016-01-12:
you can pick up the serial port but not the password, port and other related information.
I'll have another look and see if I exposed anything new with the merge of PR4107. maybe my network info will show up

andig, 2016-01-12:
Thats fine- at least the standard case would be good for my purpose at least. Got the plugin compiling meanwhile and will build the "default" upload thingy.

me-no-dev, 2016-01-12:
I'm almost done with modifications for folder support and passwordless upload to the default OTA port

me-no-dev, 2016-01-12:
https://github.com/esp8266/arduino-esp8266fs-plugin/commit/68fd73873203a78018bbcc3dd7b2f04f5976befb
I'm leaving to @igrr to get this into the release bundles :)
Keep in mind this will only work on the default OTA port and no password

igrr, 2016-01-12:
Ok, sure.

igrr, 2016-01-12:
In [0.2.0](https://github.com/esp8266/arduino-esp8266fs-plugin/releases/tag/0.2.0)

closed 2016-01-12 by igrr

attakorns, 2019-07-21:
Hmm guys..not sure if an update to the Arduino IDE allowed this but I managed to get the password from the IDE using this:
`PreferencesData.getMap().get(uploader.getAuthorizationKey());`
or if you want an exception when no password data is present:
`PreferencesData.getMap().getOrExcept(uploader.getAuthorizationKey());`

`.getMap(), .get() and .getOrExcept()` is imported by:
`import processing.app.helpers.PreferencesMap;`
`.getOrExcept()` also requires `import processing.app.helpers.PreferencesMapException;` too I think although I have not tried this.

uploader is declared as:
```
UploaderUtils uploaderInstance = new UploaderUtils();
Uploader uploader = uploaderInstance.getUploaderByPreferences(false);
```
and requires import of:
```
import cc.arduino.UploaderUtils;
import cc.arduino.packages.Uploader;
```

To check whether a password is required:
`if(uploader.requiresAuthorization())`

I also managed to get the password dialogue prompt to appear in the same fashion as Upload Sketch does too:

```
if (uploader.requiresAuthorization() && !PreferencesData.has(uploader.getAuthorizationKey())) {
  PasswordAuthorizationDialog dialog = new PasswordAuthorizationDialog(editor, tr("Type board password to upload a new sketch"));
  dialog.setLocationRelativeTo(editor);
  dialog.setVisible(true);

  if (dialog.isCancelled()) {
    editor.statusNotice(tr("Upload cancelled"));
    return;
  }

  PreferencesData.set(uploader.getAuthorizationKey(), dialog.getPassword());
} 
```
These requires:
```
import processing.app.forms.PasswordAuthorizationDialog;
import static processing.app.I18n.tr;
```

So if you do
`String password = prefs.get(uploader.getAuthorizationKey());`
you could then run espota using:
`sysExec(new String[]{pythonCmd, espota.getAbsolutePath(), "-i", serialPort, "-a", password, "-s", "-f", imagePath});`

I got these lines from the Arduino source files: GenericNetworkUploader.java and SketchController.java. I'm a total noob at Java and how Arduino IDE coding works so if this appears like a hack then it is! I'm dropping this here so someone who knows more can properly add this feature to ESP8266FS. Hope it's helpful.