esp8266/arduino-esp8266fs-plugin #5: SPIFFS data directory  subfolder not converted in prefixed files.

igrr, 2015-12-04:
_From @far5893 on December 4, 2015 9:16_

SPIFFS does not support directories so dir is only a prefix of filename,
if i prepare SPIFFS  filesystem in linux (data directory) i cannot prefix files with /somewhere/ (/ is not allowed)  but i can create directories , spiffs flash tool does not convert directories name in file prefixes!

_Copied from original issue: esp8266/Arduino#1139_

me-no-dev, 2015-12-04:
ok, what do we need to do?
get the files  in subfolders and put them with the folder name as prefix to the new name?
do the names start with "/", I think it does not matter but we need to standardize it somehow on the ESP side. I have run into issues doing SPIFFS.open("file.txt","w") and SPIFFS.open("/file.txt","w")

far5893, 2015-12-04:
In a file system where there is not working-dir concept all name should be absolute (starting with /) for compatibility with filesystem where dir is a real object.

igrr, 2015-12-04:
Actually i checked and mkspiffs 1.2 supports nested directories. So I think this should just work if you create a 'somewhere' subdirectory in 'data' directory and put all files there.

far5893, 2015-12-04:
Under linux?

me-no-dev, 2015-12-04:
@igrr where can I read on the matter?

igrr, 2015-12-04:
Recursive packing has been added here: https://github.com/igrr/mkspiffs/commit/2ecc11f76955f512ef200b59929f7a2e4acde052
It's part of mkspiffs-1.2 release.

me-no-dev, 2016-01-12:
recursive plugin have been done here: https://github.com/esp8266/arduino-esp8266fs-plugin/commit/68fd73873203a78018bbcc3dd7b2f04f5976befb

igrr, 2016-01-12:
In [0.2.0](https://github.com/esp8266/arduino-esp8266fs-plugin/releases/tag/0.2.0)

closed 2016-01-12 by igrr