esp8266/arduino-esp8266fs-plugin #7: On IDE 1.6.7: Error with uploading

nouser2013, 2016-01-03:
Greetings,

I'm on IDE 1.6.7 and it fails with the upload. I set everything up as described, and I can upload a sketch with no problems, and it seems to compile the 3 MB of file system binary, but uploading results in these messages:

```
[SPIFFS] data   : U:\ ..... \main\data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/test.txt

[SPIFFS] upload : C:\ [...] 8b9a9.spiffs/main.spiffs.bin
[SPIFFS] reset  : ck
[SPIFFS] port   : COM8
[SPIFFS] speed  : 115200
[SPIFFS] address: 0x100000

warning: espcomm_sync failed
error: espcomm_open failed
SPIFFS Upload failed!
```

Am I doing something wrong?

Thanks

jongross, 2016-01-29:
I am also having this error with Arduino 1.6.7 - any ideas on where to start?

me-no-dev, 2016-01-29:
looks like esptool problem. You do have the serial monitor closed right? You can not have it open and upload like it's done with the sketches, because we do not have access from the tool to the serial port to disconnect it.

jongross, 2016-01-29:
yeah it was closed - I just figured out the issue - the connection I have here to the chip is supposed to trigger bootload mode before doing anything but it isn't happening, so ground pin0 and rebooting the chip worked.

Thanks for the quick response.

me-no-dev, 2016-01-29:
BTW the new version of the plugin uploads over OTA as well :) you might find it easier and much faster

jongross, 2016-01-29:
I'm implementing that piece now, cheers!

kdupke, 2019-01-09:
I missed to bring my ESP-01s into flash mode before using the tool. Please add this to the documentation.