esp8266/arduino-esp8266fs-plugin #9: Version0.2.0 of "readBytes (byte, length)" I get an error at.

mgo-tec, 2016-02-01:
I will for the first time questions.
English I'm sorry the weak.

Version is 0.1.3 of the was no problem, but in ver 0.2.0 of the following sketch "readBytes (byte, length)" I get an error at.
Following message is issued.

SPIFFS_10:9: error: invalid conversion from 'byte * {aka unsigned char *}' to 'char *' [-fpermissive]

What does that mean?

I hope if you could tell me please.

My PC environment:
Windows8.1
Arduino IDE 1.6.5

Sample sketch:
#include ”FS.h”
void setup() {
  SPIFFS.begin();
  Serial.begin(115200);
  byte buf[2];

  File file = SPIFFS.open("/test.bin", "r");
  file.seek(0,SeekSet);
  file.readBytes(buf,2);
  
  Serial.println();
  for(int i=0; i<2; i++){
    Serial.println(buf[i],HEX);
  }
}
void loop() {
}

kmaximv, 2016-02-01:
Try replace  byte buf[2]; to char buf[2];

mgo-tec, 2016-02-01:
It worked when I Immediately tried in place of the char type.
I guess there's no choice but to change to the char type from the byte type in the sketch of the past.
Thank you immediately answer.

kmaximv, 2016-02-01:
You can do type conversion. But it would be worse.

mgo-tec, 2016-02-01:
That's right !
However, I have no way only do a one-by-one type conversion.
Thank you for advice.

me-no-dev, 2016-02-14:
I'm closing this issue as being unrelated to the plugin

closed 2016-02-14 by me-no-dev