esp8266/arduino-esp8266fs-plugin #10: Yet another plugin for Arduino IDE

mytrain, 2016-02-09:
Hi,

I recently developed a plugin that provide a Serial Monitor with RESET (RTS) and DTR control in order to avoid keeping the ESP in reset state when using original Arduino Serial Monitor

I posted it on https://github.com/esp8266/Arduino/issues/1360.

Now I'm looking for somebody that can help me publishing my code on GITHUB as part of this repository or another one. I do not know who should I contact to initiate this action.

Thanks

igrr, 2016-02-09:
Do you want to publish it under github.com/esp8266/arduno-serial-monitor-plugin or something different? Do you have a repository with the source code set up already?

mytrain, 2016-02-09:
github.com/esp8266/arduno-serial-monitor-plugin looks fine to me.
No i did not setup anything. I  I'm a rooky in github. Just tell me what to do.
Should I create something on my own rep https://github.com/mytrain ?