esp8266/arduino-esp8266fs-plugin #11: IllegalAccessError: plugin 0.2.0 IDE 1.6.7

sauloal, 2016-02-14:
Hello,

I've got this message:

`
Exception in thread "AWT-EventQueue-0" java.lang.IllegalAccessError: tried to access method processing.app.Sketch.prepareDataFolder()Ljava/io/File; from class com.esp8266.mkspiffs.ESP8266FS
	at com.esp8266.mkspiffs.ESP8266FS.createAndUpload(ESP8266FS.java:165)
	at com.esp8266.mkspiffs.ESP8266FS.run(ESP8266FS.java:226)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`

Any ideas?

me-no-dev, 2016-02-14:
is this a constant error condition or does it happen every time? I looked through the source and at the given lines, the above method has already been called more than once, so why it fails I have no clue...
Maybe if you give a good test scenario we can find the reason?

sauloal, 2016-02-14:
Constant.

I'm using the default FSBrowser sketch from the ESP library

sumbasu, 2016-04-13:
Hi,
I too got the same issue. How did you resolve this one?

Regards
Sumanta

technik91, 2016-05-13:
I has this problem too. I solved it update ESP8266FS to version 0.2.0