esp8266/arduino-esp8266fs-plugin #12: This plugin doesn't work with 4M(3M SPIFSS) settings but it works with 4M(1M SPIFFS)

gosewski, 2016-02-18:
I use W25Q32 Winbond memory chip in my module. I am able to upload any skech, no matter if I set 4M(3M SPIFSS) or 4M(1M SPIFFS). Unfortunately, the plugin works only with 4M(1M SPIFFS) settings. In both cases plugin compiles the bin file of a proper size 1MB and 3MB respectively but 3M settings I recive the error:

warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
SPIFFS Upload failed!

In the data folder I have only on file (for test purpose) of 17kB size. The plugin behaves identically in IDE 1.6.4 and IDE 1.6.6 environment.

ajvpot, 2016-02-19:
I have the same issue.

gosewski, 2016-02-20:
Good news!
Today I changed SPIFSS size to 2MB and I was able to load it with this plugin. The problem is not fully solved but well, 2MB SPIFFS is two times better than 1MB SPIFFS :)

I did it it in quite elegent way (I think) by creating 4M( 2M SPIFFS) entry in boards.tx:

`generic.menu.FlashSize.4M2M=4M (2M SPIFFS)
generic.menu.FlashSize.4M2M.build.flash_size=4M
generic.menu.FlashSize.4M2M.build.flash_ld=eagle.flash.4m2m.ld
generic.menu.FlashSize.4M2M.build.spiffs_start=0x1FB000
generic.menu.FlashSize.4M2M.build.spiffs_end=0x3FB000
generic.menu.FlashSize.4M2M.build.spiffs_blocksize=8192
generic.menu.FlashSize.4M2M.build.spiffs_pagesize=256
generic.menu.FlashSize.4M2M.upload.maximum_size=1044464`

and creating new loader file eagle.flash.4m2.ld:

`/* Flash Split for 4M chips */
/* sketch 1019KB */
/* spiffs 2048KB */
/* eeprom 20KB */

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xfeff0
}

PROVIDE ( _SPIFFS_start = 0x403FB000 );
PROVIDE ( _SPIFFS_end = 0x405FB000 );
PROVIDE ( _SPIFFS_page = 0x100 );
PROVIDE ( _SPIFFS_block = 0x2000 );

INCLUDE "../ld/eagle.app.v6.common.ld"`

Tomorrow I will contiue my investigation ;)

gosewski, 2016-02-20:
Ok, I didn't find the reason but I found the solutions and now I have 3MB SPIFFS.
To do that one must modity boards.txt file:
`
generic.menu.FlashSize.4M3M.build.spiffs_start=0x109000
`
and eagle.flash.4m.ld file:
`
PROVIDE ( _SPIFFS_start = 0x40309000 );
`
So now I am able to serve a lot of graphics from SPIFFS on my dynamically created ESP8266 web page.

If one needs 4KB extra SPIFFS then one needs to reduce the spiff blosck size and go down with spiffs start address:
boards.txt file:
`
generic.menu.FlashSize.4M3M.build.spiffs_start=0x108000
generic.menu.FlashSize.4M3M.build.spiffs_blocksize=4096
`
eagle.flash.4m.ld file:
`
PROVIDE ( _SPIFFS_start = 0x40308000 );
PROVIDE ( _SPIFFS_block = 0x1000 );
`

ajvpot, 2016-02-20:
Also, have you noticed that SPIFFS is very slow? Serving about 1mb takes me 20 seconds on my board.

gosewski, 2016-02-23:
Could you please show the code you use for serving this file(s)? What is the spiffs_blocksize in your board configuration? What is the flash and CPU frequency of your board?

ajvpot, 2016-02-23:
The code is borrowed from https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino . spiffs_blocksize=8192, cpu freq 80MHz or 160MHz doesn't make a difference.

It's a NodeMCU 0.9 Devboard.

gosewski, 2016-02-23:
ok, thank you.
You didn't mentioned the flash speed, is it 40 or 80MHz?
I did some quick test, my parameters and results are as follows:
PARAMS:
CPU Speed: 160MHz
Flash Speed: 80MHz QIO
SPIFFS: 3MB, block size 8KB
File size: 1MB (1 097 526 B) 
Write buffer: 1460B

RESULTS:
/test.png: 3591 ms  @ -64 dBm
/test.png: 4448 ms  @ -74 dBm

So, in my case serving 1MB file from SPIFFS to browser takes about 3.5s when the WiFi signal strength is 72% and 4.5s when the signal strength is 52%

For tests I used plain http server, here  is the code:

```
File f= SPIFFS.open("test.png", "r");
client.print("HTTP/1.1 200 OK\nContent-Length: "+f.size()+(String)"Content-Type: image/png\n\n");
Serial.print(f.name());
Serial.print(": ");
long s=millis();
client.write(f,1460);
s=millis()-s;
Serial.print(s);
Serial.print(" ms  @");
Serial.print(WiFi.RSSI());
Serial.print(" dBm");
f.close();
```

nopnop2002, 2017-01-23:
I had the same problem, too.
I exchanged USB-TTL converter from PL-2303 to CH340 .
A problem was resolved.