esp8266/arduino-esp8266fs-plugin #13: SPIFFS hangs during uploading?

Rickgg, 2016-02-19:
When I try to flash the files, the SPIFFS uploader "hangs" and stops transmitting. I'm using Arduino IDE 1.6.5 and Mac OS X Yosemite. Any ideas as to why this could be happening?

me-no-dev, 2016-02-23:
what USB to UART are you guys using? I had tons of problems with FTDI
what size SPIFFS?

kaeferfreund, 2016-02-23:
64k SPIFFS with FTDI converter. But the FTDI works quite fine with everything else...
Trying a CH340 later on

me-no-dev, 2016-02-23:
issue with ftdi comes when it transmits a lot of 0xFF which is EMPTY byte in the flash, because the signal  looks constant and it thinks that the connection failed. For the sketch a patch was added to esptool to pad the firmware (the space between the bootloader and the firmware) with 0x55 instead to transmit pulses and not fail sketch upload.

me-no-dev, 2016-02-23:
You can move to 1.5.7+ and use OTA to upload SPIFFS images much faster and reliable :)
The issue seem to not exist on ElCapitan but the FTDI needs to be behind USB hub in order to work (changes in the USB stack from Apple broke some drivers)

kaeferfreund, 2016-02-23:
thx me-no-dev, just waiting for my esp-12 to arrive, bec. the esp-01 doesn't have enough space for ota updating.

Rickgg, 2016-03-07:
How do you upload SPIFFS files through OTA? I didn't find any instructions in the docs.

Rickgg, 2016-03-07:
Found it! should have looked a little bit harder... hehehe :grin:

closed 2016-03-07 by Rickgg