esp8266/arduino-esp8266fs-plugin #14: 4M (2M SPIFFS) - possible?

jlippa, 2016-03-16:
Not really an issue, just a question. Is it possible to split flash in half where 50% is used for firmware code and 50% is used for SPIFFS? 

I see the definitions defined in boards.txt but I'm not sure if to get this working, this is a case of just creating a new board profile using the eagle.flash.4m.ld and specifying a new SPIFFS start address or if a brand new eagle.flash.* layout would be required.

me-no-dev, 2016-03-16:
the ESP maps only 1MB of the flash and that is the reason such setting does not exist :) Only 1MB can be used for sketch

igrr, 2016-03-16:
It's possible to do that, but you won't be able to run code which is above 1M mark. So this may be useful for OTA updates if the size of your firmware exceeds 512k, but I haven't seen any project which approaches this size yet.
If you do that, then yes, you need to create a linker script variant and add an entry to boards.txt.

jlippa, 2016-03-16:
Running out of free space for OTA update was one of the things I was thinking about..but it's a concern that's probably some way off because my .bin is currently 312,168 bytes when built.

Learning about how to create a new linker script variant .ld sounds interesting though. Do you know if there's any info about how to do this anywhere?

casperx102, 2017-01-08:
how about this? http://wiki.jackslab.org/images/Esp8266-flash-layout-ota-4096.png

Pablo2048, 2017-12-14:
Hi, please see the attached linker script for 2M/4M variant. Also there is need to change boards.txt (now probably by some python script) with this:
[eagle.flash.4m2m.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/1558912/eagle.flash.4m2m.zip)

 linker script for 2M/4M variant