esp8266/arduino-esp8266fs-plugin #17: How to set up tool for OTA update?

viktak, 2016-04-20:
I just installed the latest Arduino IDE, the "ESP-extension" for it, then copied this tool to the folder that is specified in the readme. The tool shows up in the Tools menu, but there is no option to select anything related to OTA, see screenshot below:
[image]
I'm developing for an ESP-12 board, then environment is Windows 10 Pro 64.

What am I missing?

Thanks for any help in advance!

me-no-dev, 2016-04-20:
when you upload an OTA enabled sketch to your ESP, the device will show in the Port menu.
So, when you have selected a serial port, it will upload using the serial port, else it will try OTA

viktak, 2016-04-20:
@me-no-dev I don't have a serial port (I do development on another machine), but I do have an OTA enabled firmware in the ESP. I use platformio with the Arduino framework for development, and each update goes smoothly to the ESP.

me-no-dev, 2016-04-20:
if your OTA is working fine, then the ESP must be visible and selectable in the Port menu of you screenshot. Select the board and upload

viktak, 2016-04-20:
@me-no-dev Well, does the OTA have to have some specific bits? And I still don't understand: if I don't specify an IP address, how does it know where to upload the files? what if I have multiple ESPs on my network? I still think I'm missing something, something basic, that is obvious to you, but me, as a newcomer to this tool may overlook.

me-no-dev, 2016-04-21:
you said you have platformio setup with OTA and you upload successfully. How do you upload? How is your firmware OTA enabled? you can see instructions [here](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#arduino-ide)