esp8266/arduino-esp8266fs-plugin #19: "Invalid library found" with Arduino IDE 1.6.9 and arduino-esp8266fs-plugin 0.2.0

HorstBaerbel, 2016-05-12:
I'm using the 64bit variant of the Arduino IDE if that matters.

me-no-dev, 2016-06-02:
isn't this error coming from Arduino IDE and the libraries that you have installed?

HorstBaerbel, 2016-06-02:
Yes. specifically the arduino-esp8266fs-plugin library aka esp8266fs.jar...

me-no-dev, 2016-06-03:
we had another report similar to what you are saying. Building the plugin from source helped.
What OS are you on?

HorstBaerbel, 2016-06-04:
Ubuntu 14.04.03 x64

me-no-dev, 2016-06-04:
ok so that seems common... the other user also on ubuntu64 Arduino IDE 1.6.9
How should I install the IDE to reproduce this issue?
You can read here on the workaround: https://github.com/esp8266/arduino-esp8266fs-plugin/issues/20

HorstBaerbel, 2016-06-04:
As the 1.6.9 ide was not available via the package manager I downloaded it and installed it following [this guide](https://www.arduino.cc/en/Guide/Linux). Then I copied the FS library to the library directory following the [README](https://github.com/esp8266/arduino-esp8266fs-plugin/).
Will check out the workaround.

HorstBaerbel, 2016-06-04:
I seemed to have no java DK installed, as make.sh told me "./make.sh: Line 22: javac: Command not found". After "sudo apt-get install default-jdk" it tells me:

> pde_path: ../../../opt/arduino-1.6.9/lib/pde.jar
> ../../../usr/share/arduino-1.5.8/lib/pde.jar
> core_path: ../../../opt/arduino-1.6.9/lib/arduino-core.jar
> lib_path: ../../../opt/arduino-1.6.9/lib/commons-codec-1.7.jar
> ../../../usr/share/arduino-1.5.8/lib/commons-codec-1.7.jar
> javac: invalid target release: 1.8
Could the older Arduino IDE install be causing the issue? How would I get rid of it?

me-no-dev, 2016-06-05:
it seems that the default java version on ubuntu64 is 1.7(OpenJDK-7) while on all other OSs is 1.8(OpenJDK-8).
edit [this](https://github.com/esp8266/arduino-esp8266fs-plugin/blob/master/make.sh#L22) line to replace 1.8 with 1.7 and give it another go?

HorstBaerbel, 2016-06-05:
> pde_path: ../../../opt/arduino-1.6.9/lib/pde.jar
> ../../../usr/share/arduino-1.5.8/lib/pde.jar
> core_path: ../../../opt/arduino-1.6.9/lib/arduino-core.jar
> lib_path: ../../../opt/arduino-1.6.9/lib/commons-codec-1.7.jar
> ../../../usr/share/arduino-1.5.8/lib/commons-codec-1.7.jar
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(processing/app/PreferencesData.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> src/ESP8266FS.java:38: error: cannot find symbol
> import processing.app.Editor;
>                      ^
>   symbol:   class Editor
>   location: package processing.app
> src/ESP8266FS.java:39: error: cannot find symbol
> import processing.app.Base;
>                      ^
>   symbol:   class Base
>   location: package processing.app
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(processing/app/BaseNoGui.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(processing/app/Platform.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> src/ESP8266FS.java:42: error: cannot find symbol
> import processing.app.Sketch;
>                      ^
>   symbol:   class Sketch
>   location: package processing.app
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(processing/app/SketchData.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> src/ESP8266FS.java:44: error: cannot find symbol
> import processing.app.tools.Tool;
>                            ^
>   symbol:   class Tool
>   location: package processing.app.tools
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(processing/app/helpers/ProcessUtils.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(processing/app/debug/TargetPlatform.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> src/ESP8266FS.java:48: error: package org.apache.commons.codec.digest does not exist
> import org.apache.commons.codec.digest.DigestUtils;
>                                       ^
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(processing/app/helpers/FileUtils.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(cc/arduino/files/DeleteFilesOnShutdown.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> src/ESP8266FS.java:56: error: cannot find symbol
> public class ESP8266FS implements Tool {
>                                   ^
>   symbol: class Tool
> src/ESP8266FS.java:57: error: cannot find symbol
>   Editor editor;
>   ^
>   symbol:   class Editor
>   location: class ESP8266FS
> src/ESP8266FS.java:60: error: cannot find symbol
>   public void init(Editor editor) {
>                    ^
>   symbol:   class Editor
>   location: class ESP8266FS
> src/ESP8266FS.java:114: error: cannot find symbol
>   private String getBuildFolderPath(Sketch s) {
>                                     ^
>   symbol:   class Sketch
>   location: class ESP8266FS
> src/ESP8266FS.java:116: error: cannot find symbol
>       File buildFolder = FileUtils.createTempFolder("build", DigestUtils.md5Hex(s.getMainFilePath()) + ".spiffs");
>                                                              ^
>   symbol:   variable DigestUtils
>   location: class ESP8266FS
> warning: ../../../opt/arduino-1.6.9/lib/arduino-core.jar(processing/app/helpers/PreferencesMap.class): major version 52 is newer than 51, the highest major version supported by this compiler.
>   It is recommended that the compiler be upgraded.
> 10 errors
> 9 warnings

Diluxshan, 2018-10-12:
You have to delete the alter folder inside it...........
and need to put the correct .h file to run the code nodemcu