esp8266/arduino-esp8266fs-plugin #20: Plugin Fails to Load on Arduino 1.6.9 Ubuntu Gnome 16.06

ehughes, 2016-06-01:
I am running Ubuntu Gnome in a VM (16.06).   When I copy the plugin into Tools/ESP8266FS/tool/ESP8266FS.jar and restart the Arduino IDE,  I do not get the extra command in the tools menu.

I tested on a Windows 10 machine and it works OK.

me-no-dev, 2016-06-02:
can you try to build it yourself? in your Arduino installation folder there should be "tools" folder. Clone the repo inside and sun the build script. it should build and put the jar in the proper folder. Maybe some Java related issue...

ehughes, 2016-06-03:
I was able to rebuild the plugin.

The build script copies the plugin to ~/Documents/Arduino/

I found that I had to copy the tools/ESP8266FS.....   to ~/Arduino

After performed the build and copied to that location it seemed to work OK.   

Lastly,  I did have to recompile for it to work.   

This is a fresh install of Ubuntu 16.06 on a VMware with Arduino 1.6.9.    Maybe something change in the last version of where it looks for things.

closed 2016-06-03 by ehughes