esp8266/arduino-esp8266fs-plugin #22: ERROR: AWT-EventQueue-0   Using arduino Ver1.6.11   ESP8266  Sketch Data Upload

k31ww475, 2016-10-09:
Hi.
I want to use " ESP8266  Sketch Data Upload "  but I couldn't.
The upload file is sample.txt. The file contents is just "abc".
I have a trouble.Thanks in advance.

THE ERROR :

Exception in thread "AWT-EventQueue-0" java.lang.IllegalAccessError: tried to access method processing.app.Sketch.prepareDataFolder()Ljava/io/File; from class com.esp8266.mkspiffs.ESP8266FS
	at com.esp8266.mkspiffs.ESP8266FS.createAndUpload(ESP8266FS.java:165)
	at com.esp8266.mkspiffs.ESP8266FS.run(ESP8266FS.java:226)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

k31ww475, 2016-10-09:
Arduino ESP8266 filesystem uploader ver. is Latest

eketjall, 2016-11-28:
I get similar errors with 0.3.0 and Arduino IDE 1.6.11 & 1.6.13
`Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
   at com.esp8266.mkspiffs.ESP8266FS.getBuildFolderPath(ESP8266FS.java:117)
   at com.esp8266.mkspiffs.ESP8266FS.createAndUpload(ESP8266FS.java:263)
   at com.esp8266.mkspiffs.ESP8266FS.run(ESP8266FS.java:322)
   at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
   at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
   at java.awt.EventQueue.access$500(EventQueue.java:97)
   at java.awt.EventQueue$3.run(EventQueue.java:709)
   at java.awt.EventQueue$3.run(EventQueue.java:703)
   at java.security.AccessController.doPrivileged(Native Method)
   at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
   at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
   at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
   at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
   at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
   at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`
0.2.0 works fine though

me-no-dev, 2016-11-28:
hmmm something must be wrong. I use 1.6.12 and all is fine. As a matter of fact that change was introduced by a member of the Arduino team and is exactly to make the plugin work in latest IDEs.
Make sure you have only one IDE running and that it's preferable 1.6.13. Make sure you restart the IDE and have only one plugin and so on.

eketjall, 2016-11-30:
Yeah, something seems broken alright. 
I follow the instruktions for installing, same way for both 0.2.0 and 0.30. Both in IDE 1.6.11 and 1.6.13.
The result is the same, 0.2.0 works in both 1.6.11 and 1.6.13. 
0.3.0 works in neither. 
I have one more plugin though, the Modified Serial Monitor, https://github.com/esp8266/Arduino/issues/1360

hawkagent, 2016-12-04:
I'm using Arduino 1.6.15 and 0.30 gives me the same error
Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File; 

0.20 works for me.

regimantas, 2016-12-19:
Hello, I'm using Arduino 1.6.9 and my error looks same:

Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.esp8266.mkspiffs.ESP8266FS.getBuildFolderPath(ESP8266FS.java:117)
	at com.esp8266.mkspiffs.ESP8266FS.createAndUpload(ESP8266FS.java:263)
	at com.esp8266.mkspiffs.ESP8266FS.run(ESP8266FS.java:322)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

regimantas, 2016-12-19:
My OS is windows10 and java version:
java version "1.8.0_111"
Java(TM) SE Runtime Environment (build 1.8.0_111-b14)
Java HotSpot(TM) 64-Bit Server VM (build 25.111-b14, mixed mode)

bjpirt, 2017-12-19:
In case anyone else experiences this issue. I had this happening on IDE v1.6.7 but updated and it runs fine on the latest (v1.8.5 currently)

girishsukukumar, 2019-09-21:
EVen I had faced the same issue, I just moved to a more recent version which I had downloaded and installed (Which is 1.8.9) and this error went way and Upload SPIFFS worked well.