esp8266/arduino-esp8266fs-plugin #24: Upload fails on certain USB-TTL adapters

brahmafear, 2016-12-22:
Plugin works for me when using certain USB-TTL adapters but fails (freezes progress) on certain models.  Posting here in case anyone has seen same issue.  Tested with two models of problem

 adapter listed below - both show same issue.  (Both work perfectly when uploading sketch from IDE.)  

Mac OS X 10.11.6
Arduino IDE 1.6.12
USB-TTL prodID 6001, vendID 0403

Note: this USB-TTL adapter is supposedly an FTDI chip, but since I got them for cheap online, I'm open to the possibility that they are clones.  However, posting issue in case this is indicative of a problem in the plugin code.  

Attached file shows where progress in IDE upload freezes.  This is running the upload plugin on the FSBrowser example sketch in ESPWebServer package.

[image]

brahmafear, 2016-12-22:
More info:  only way to "unfreeze" the IDE is to disconnect the USB connection to the USB-TTL adapter.  Upon doing so, the following is shown in the IDE:

Uploading 131072 bytes from /var/folders/nb/dtcb7q0555nbdbjgslnf64fr0000gp/T/arduino_build_935934/FSBrowser.spiffs.bin to flash at 0x0005B000
...................................Error while setting serial port parameters: 115,200 N 8 1
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 4 bytes
error: failed sending 0xC0
warning: espcomm_send_command: didn't receive command response
error: espcomm_upload_mem failed
SPIFFS Upload failed!

me-no-dev, 2016-12-22:
this looks like FTDI adapter. See to update the driver and preferably have it connected through USB hub. It's something to do with USB on Mac

brahmafear, 2016-12-22:
Very possible.  In either case, I can work around by using a different adapter.  Will mark issue as closed.  The thread will remain searchable in case it happens to anyone else in future.  Thanks.

closed 2016-12-22 by brahmafear

brahmafear, 2016-12-23:
More info for future:
Installed official FTDI v2.3 driver for OSX instead of using OSX built-in driver for FTDI:  same issue, same behavior.  Will test with USB hub when possible.  Will blame issue on shady FTDI clone chip.