esp8266/arduino-esp8266fs-plugin #28: Use from command line?

ShockDog, 2017-02-08:
Is it possible to use this tool from the command line (Windows)? Maybe it is possible via Arduino-Builder, or as a standalone process. Either would suit.

me-no-dev, 2017-02-08:
it uses mkspiffs and espota/serial upload underneath so yes it is possible. The plugin reads info from the selected board to determine the size and other properties of the SPIFFS image, so you need to know those beforehand if you want to do it manually.
Magic happens [here](https://github.com/esp8266/arduino-esp8266fs-plugin/blob/master/src/ESP8266FS.java#L280-L318)

lrmoreno007, 2018-01-04:
Hi! It would be great to have a tool that works under the command line and can upload one or more files or the entire "data" folder. Through python it would be multiplatform and would be the icing on the cake.

Currently I upload the data files (TXT / CNC) through the FSbrowser web (ESPAsyncWebServer).

I've been trying to find something like this all day and I can not find it anywhere and unfortunately I think that even if I was born 5 times, I would never make it.

Regards.

beikeland, 2018-01-04:
Uploading single files would certainly be easiest with wget/curl and a webserver. Using espota.py requires the whole SPIFFS image to be built and sent which will overwrite existing files, takes much longer and (for now?) forces a reboot of the ESP8266 after espota.py is done.

lrmoreno007, 2018-01-04:
Yes, but a command line would allow to integrate the upload of files with other programs or languages, as for example in my case with AutoLisp and VBA in AutoCAD.

Regards.

igrr, 2018-01-04:
Filesystem binary generation is done using mkspiffs, and uploading can be done using esptool-ck or esptool.py.

lrmoreno007, 2018-01-04:
But, how to upload only one file to filesystem with a command line? Or multiple files?

beikeland, 2018-01-04:
As I said, keep the webserver on the ESP8266, and use curl as your command line tool to send the file. Easiest might be to use the network console in chrome (press F12, go to Network tab) to capture the network traffic while uploading a file to the ESP8266 and then right click the request get a curl command as a starting point.
[image]

Doesn't really sound like you want to generate and flash a new SPIFFS, so stick with the webserver and use curl as your command line tool.

lrmoreno007, 2018-01-04:
I understand what you want to tell me, but unfortunately I do not have enough knowledge in http, curl, etc. and it would be impossible for me.

Luckily I managed to do something similar to what I wanted through the esp8266FTPServer library of nailbuster and Filezilla.

I still can not upload the files using a command line, but in a simple and fast way I can make a bidirectional communication of files. It is enough for me although I will continue investigating other methods.

Regards and thanks.

beikeland, 2018-01-04:
Give curl a go, availbale for win/mac/linux - great tool. Supports ftp
uploads as well, so should solve your problem for now.

# curl -T file-to-upload.ext  ftp://username:password@ftp.server.com/path/

https://curl.haxx.se/download.html or your OS package manager.

lrmoreno007, 2018-01-11:
WOW !! thank you very much for the trick! I did not know this program and I thought it was harder to use.

It works perfectly!! Now using a bat file in Windows I can upload one or more files of the same type that exist in a directory and then if everything is correct move them to another directory.

Send.bat
```
@echo off
forfiles /m *.iso /c "cmd /c curl -T @file ftp://admin:admin@cncserver"
IF %ERRORLEVEL% EQU 0 (
  move /y *.iso Store
)
```
Easy and effective. Thank you very much!