esp8266/arduino-esp8266fs-plugin #31: Esp8266fs tool dont visible in menu after update on Arduino 1.8.2

pilnikov, 2017-04-06:
Arduino 1.8.2 banned Esp8266fs tool in menu. In 1.8.1 all work.

DarkiStar, 2017-04-08:
for me it works without problems in the 1.8.2

pilnikov, 2017-04-08:
version your Windows? My is 10x64 prof

DarkiStar, 2017-04-09:
Same Windows Version on my side.

pilnikov, 2017-04-09:
Please show your path to the plugin

DarkiStar, 2017-04-09:
D:\Downloads\arduino-1.8.2\tools\ESP8266FS\tool\esp8266fs.jar

DarkiStar, 2017-04-11:
Would like to know if it works for you now ...

pilnikov, 2017-04-11:
No, it does not work. I checked all the paths, updated the version to 0.3.0. Did not help.

DarkiStar, 2017-04-11:
And you have selected an ESP8266 Board? Just to be sure ...
[image]

pilnikov, 2017-04-11:
[image]

pilnikov, 2017-04-11:
[image]

pilnikov, 2017-04-11:
I repaired everything! Has earned !!!! Was c:\Program Files (x86)\Arduino\tools\ESP8266FS\tool**s**\  ... It is necessary so c:\Program Files (x86)\Arduino\tools\ESP8266FS\tool**\**

closed 2017-04-11 by pilnikov