esp8266/arduino-esp8266fs-plugin #35: file "esp8266fs.jar" does not exist

shiftleftplusone, 2017-06-06:
The file "esp8266fs.jar" mentioned here

> 
Installation

    Make sure you use one of the supported versions of Arduino IDE and have ESP8266 core installed.
    Download the tool archive from releases page.
    In your Arduino sketchbook directory, create tools directory if it doesn't exist yet.
    Unpack the tool into tools directory (the path will look like <home_dir>/Arduino/tools/ESP8266FS/tool/esp8266fs.jar).
    Restart Arduino IDE.

> 

does not exist, just  "esp8266fs.java", but putting it into my sketchbook folder
<home_dir>/Arduino/tools/ESP8266FS/tool/esp8266fs.java
does not show up then an Arduino tools option "ESP8266 Sketch Data Upload".

what is going wrong?

me-no-dev, 2017-06-07:
OS? maybe <homedir>/Arduino is elsewhere?

shiftleftplusone, 2017-06-07:
it's in the sketchbook folder
D:\Arduino\patch_libs\Arduino
with the subdirectories
D:\Arduino\patch_libs\Arduino\examples
D:\Arduino\patch_libs\Arduino\libraries
D:\Arduino\patch_libs\Arduino\tools
and herein now also
D:\Arduino\patch_libs\Arduino\tools\ESP8266FS\tool

but also when putting the java file into the system installation directory
C:\Program Files (x86)\Arduino\tools
it does not work.

And why is there just a .java file and not a .jar file??

shiftleftplusone, 2017-06-07:
t's in the sketchbook folder
D:\Arduino\patch_libs\Arduino
with the subdirectories
D:\Arduino\patch_libs\Arduino\examples
D:\Arduino\patch_libs\Arduino\libraries
D:\Arduino\patch_libs\Arduino\tools
and herein now also
D:\Arduino\patch_libs\Arduino\tools\ESP8266FS\tool

but also when putting the java file into the system installation directory
C:\Program Files (x86)\Arduino\tools
it does not work.

And why is there just a .java file and not a .jar file??

igrr, 2017-06-07:
@shiftleftplusone um... i think you are downloading the source code of this repository, instead of downloading a release (binary version). Double check what you are downloading exactly.

me-no-dev, 2017-06-07:
AAhh I totally missed the .java in your first message! See here :) https://github.com/esp8266/arduino-esp8266fs-plugin/releases

shiftleftplusone, 2017-06-07:
AAhh 
YEP!

now with that it works!
 https://github.com/esp8266/arduino-esp8266fs-plugin/releases

perhaps the repo should be updated! 8)

thanks a lot!

me-no-dev, 2017-06-07:
no, you should read more carefully the instructions ;) it's clearly linked to the releases

closed 2017-06-07 by me-no-dev

shiftleftplusone, 2017-06-07:
no, if there is a button  [Clone or Download]  then I expect this button to work, and if a jar file required as described then I expect this jar file to be in there, not a java file that does nor work.

me-no-dev, 2017-06-07:
You are obviously new to github :) that button clones the repository (source) or downloads and archive of the current source. Releases is where you find compiled stuff and that is clearly noted in the repository [README](https://github.com/esp8266/arduino-esp8266fs-plugin/blob/master/README.md#installation)

shiftleftplusone, 2017-06-08:
you are right, I am quite new to github although most Arduino-related github sites have a button or link to download a zip file, in order to use the Arduino function to install a lib by a zip file . OTOH, they (you?) talk about "Download the tool archive from releases page.": What the heck is a "tool archieve" and what di I have to do with releases pages instead? That sounds as if it is just for developers for obscured things like forks and all about that is absolutely not clear at all for hobby users.

TangiblePS, 2019-02-25:
I had this problem too - thanks shiftleftplusone for asking and me-no-dev for answering.

In my case the confusion was caused by the link in the instructions only showing the 'latest' release, which right now does not include the binaries, only the source code.

I'd suggest updating the link in the instructions to not include /latest .

ed-french, 2019-02-25:
I had the same problem. The Readme for the repo links to the wrong url I think:
https://github.com/esp8266/arduino-esp8266fs-plugin/releases/tag/0.4.0
Is not correct?