esp8266/arduino-esp8266fs-plugin #38: How to compile JAR ? Compatibility with ESP32 ?

O19E4, 2017-10-21:
Hi.
I planed to edit the ESP8266FS.java according to https://github.com/espressif/arduino-esp32/issues/411, copercini 17 July.
I installed "jdk-8u152-windows-x64.exe" and, starting a bash GUI (providen with GIT installation) from "C:\Program Files (x86)\Arduino\tools\ESP8266FS\tool", I got following errors (see below).

1. I was not able to fix this. I download from "https://playground.arduino.cc/Interfacing/Processing" and follow the instructions, but it seems there is no relevant JAR to this (maybe I missed something ?)
2. Does your work (ESP8266FS.jar) will be compatible with ESP32 ? When ?

Thanks in advance ;)

Regards.

```
src\ESP8266FS.java:37: error: package processing.app does not exist
import processing.app.PreferencesData;
                     ^
src\ESP8266FS.java:38: error: package processing.app does not exist
import processing.app.Editor;
                     ^
src\ESP8266FS.java:39: error: package processing.app does not exist
import processing.app.Base;
                     ^
src\ESP8266FS.java:40: error: package processing.app does not exist
import processing.app.BaseNoGui;
                     ^
src\ESP8266FS.java:41: error: package processing.app does not exist
import processing.app.Platform;
                     ^
src\ESP8266FS.java:42: error: package processing.app does not exist
import processing.app.Sketch;
                     ^
src\ESP8266FS.java:43: error: package processing.app does not exist
import processing.app.SketchData;
                     ^
src\ESP8266FS.java:44: error: package processing.app.tools does not exist
import processing.app.tools.Tool;
                           ^
src\ESP8266FS.java:45: error: package processing.app.helpers does not exist
import processing.app.helpers.ProcessUtils;
                             ^
src\ESP8266FS.java:46: error: package processing.app.debug does not exist
import processing.app.debug.TargetPlatform;
                           ^
src\ESP8266FS.java:49: error: package processing.app.helpers does not exist
import processing.app.helpers.FileUtils;
                             ^
src\ESP8266FS.java:51: error: package cc.arduino.files does not exist
import cc.arduino.files.DeleteFilesOnShutdown;
                       ^
src\ESP8266FS.java:56: error: cannot find symbol
public class ESP8266FS implements Tool {
                                  ^
  symbol: class Tool
src\ESP8266FS.java:57: error: cannot find symbol
  Editor editor;
  ^
  symbol:   class Editor
  location: class ESP8266FS
src\ESP8266FS.java:60: error: cannot find symbol
  public void init(Editor editor) {
                   ^
  symbol:   class Editor
  location: class ESP8266FS
src\ESP8266FS.java:114: error: cannot find symbol
  private String getBuildFolderPath(Sketch s) {
                                    ^
  symbol:   class Sketch
  location: class ESP8266FS
src\ESP8266FS.java:71: error: cannot find symbol
        final Process p = ProcessUtils.exec(arguments);
                          ^
  symbol:   variable ProcessUtils
  location: class ESP8266FS
src\ESP8266FS.java:126: error: cannot find symbol
        File buildFolder = FileUtils.createTempFolder("build", DigestUtils.md5Hex(s.getMainFilePath()) + ".tmp");
                           ^
  symbol:   variable FileUtils
  location: class ESP8266FS
src\ESP8266FS.java:137: error: cannot find symbol
          method = BaseNoGui.class.getMethod("getBuildFolder");
                   ^
  symbol:   class BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:155: error: cannot find symbol
    String data = BaseNoGui.getBoardPreferences().get(name);
                  ^
  symbol:   variable BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:162: error: cannot find symbol
    if(!PreferencesData.get("target_platform").contentEquals("esp8266")){
        ^
  symbol:   variable PreferencesData
  location: class ESP8266FS
src\ESP8266FS.java:164: error: cannot find symbol
      editor.statusError("SPIFFS Not Supported on "+PreferencesData.get("target_platform"));
                                                    ^
  symbol:   variable PreferencesData
  location: class ESP8266FS
src\ESP8266FS.java:168: error: cannot find symbol
    if(!BaseNoGui.getBoardPreferences().containsKey("build.spiffs_start") || !BaseNoGui.getBoardPreferences().containsKey("build.spiffs_end")){
        ^
  symbol:   variable BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:168: error: cannot find symbol
    if(!BaseNoGui.getBoardPreferences().containsKey("build.spiffs_start") || !BaseNoGui.getBoardPreferences().containsKey("build.spiffs_end")){
                                                                              ^
  symbol:   variable BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:170: error: cannot find symbol
      editor.statusError("SPIFFS Not Defined for "+BaseNoGui.getBoardPreferences().get("name"));
                                                   ^
  symbol:   variable BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:186: error: cannot find symbol
    TargetPlatform platform = BaseNoGui.getTargetPlatform();
    ^
  symbol:   class TargetPlatform
  location: class ESP8266FS
src\ESP8266FS.java:186: error: cannot find symbol
    TargetPlatform platform = BaseNoGui.getTargetPlatform();
                              ^
  symbol:   variable BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:190: error: cannot find symbol
    if(PreferencesData.get("runtime.os").contentEquals("windows"))
       ^
  symbol:   variable PreferencesData
  location: class ESP8266FS
src\ESP8266FS.java:199: error: cannot find symbol
        tool = new File(PreferencesData.get("runtime.tools.mkspiffs.path"), mkspiffsCmd);
                        ^
  symbol:   variable PreferencesData
  location: class ESP8266FS
src\ESP8266FS.java:211: error: cannot find symbol
    String serialPort = PreferencesData.get("serial.port");
                        ^
  symbol:   variable PreferencesData
  location: class ESP8266FS
src\ESP8266FS.java:236: error: cannot find symbol
          esptool = new File(PreferencesData.get("runtime.tools.esptool.path"), esptoolCmd);
                             ^
  symbol:   variable PreferencesData
  location: class ESP8266FS
src\ESP8266FS.java:265: error: cannot find symbol
    String resetMethod = BaseNoGui.getBoardPreferences().get("upload.resetmethod");
                         ^
  symbol:   variable BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:266: error: cannot find symbol
    String uploadSpeed = BaseNoGui.getBoardPreferences().get("upload.speed");
                         ^
  symbol:   variable BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:267: error: cannot find symbol
    String uploadAddress = BaseNoGui.getBoardPreferences().get("build.spiffs_start");
                           ^
  symbol:   variable BaseNoGui
  location: class ESP8266FS
src\ESP8266FS.java:304: error: cannot find symbol
      if(PreferencesData.get("runtime.os").contentEquals("windows"))
         ^
  symbol:   variable PreferencesData
  location: class ESP8266FS

```

me-no-dev, 2017-10-23:
ESP32 plugin here: https://github.com/me-no-dev/arduino-esp32fs-plugin

This is built using Arduino and not Processing :)

closed 2017-10-23 by me-no-dev

O19E4, 2017-10-23:
Hi. Thanks for the link. But I still would like to be able to compile (myself) the .java to .jar. I got quite the same errors. Could you tell me the exacts steps to compile/build the .java to .jar ? This would be great. Thanks in advance.

Best Regards.

igrr, 2017-10-23:
Please have a look at travis-ci build script in this repository, it has
exact steps.

attakorns, 2019-07-21:
I got this same problem. The problem is you have to tell javac where those classes are. So I checked out the [make.sh](https://github.com/esp8266/arduino-esp8266fs-plugin/blob/master/make.sh) file, and saw on line 22 that that's the command you need for javac. Adapting to Windows and got this to work by running:
javac -target 1.8 -cp "C:\Program Files (x86)\Arduino\lib\pde.jar;C:\Program Files (x86)\Arduino\lib\arduino-core.jar;C:\Program Files (x86)\Arduino\lib\commons-codec-1.7.jar" -d bin src/ESP8266FS.java
from root directory (so the source code is in the src directory above this directory). This of course depends on where your Arduino installation is. 
Hopefully this is useful to someone!

gotnull, 2023-01-06:
I get the following: 

```
D:\Development\arduino-esp32fs-plugin>javac -target 1.8 -cp "d:\Program Files (x86)\Arduino\lib\pde.jar;d:\Program Files (x86)\Arduino\lib\arduino-core.jar;d:\Program Files (x86)\Arduino\lib\commons-codec-1.7.jar" -d bin .\src\ESP32FS.java
warning: target release 1.8 conflicts with default source release 17
```