esp8266/arduino-esp8266fs-plugin #39: src/ESP8266FS.java:117: error: cannot find symbol

max5555, 2017-11-21:
I failed to find esp8266fs.jar, so I tried to compile it, but I've got the following error. 

```
~/Arduino/tools/arduino-esp8266fs-plugin-master$ bash make.sh
 INSTALLDIR: /home/y/Documents/Arduino
pde_path: ../../../.local/share/umake/ide/arduino/lib/pde.jar
core_path: ../../../.local/share/umake/ide/arduino/lib/arduino-core.jar
lib_path: ../../../.local/share/umake/ide/arduino/lib/commons-codec-1.7.jar
src/ESP8266FS.java:117: error: cannot find symbol
      String buildpath = s.getBuildPath().getAbsolutePath();
                          ^
  symbol:   method getBuildPath()
  location: variable s of type Sketch
1 error
```

Ubuntu 16.04x64

max5555, 2017-11-22:
I have found compiled `esp8266fs.jar` here https://github.com/esp8266/arduino-esp8266fs-plugin/releases/latest

But I still get the error on line 117:

```
Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.esp8266.mkspiffs.ESP8266FS.getBuildFolderPath(ESP8266FS.java:117)
	at com.esp8266.mkspiffs.ESP8266FS.createAndUpload(ESP8266FS.java:263)
	at com.esp8266.mkspiffs.ESP8266FS.run(ESP8266FS.java:322)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

Arduino IDE 1.6.11

me-no-dev, 2017-11-23:
too old IDE ;) you need something newer. Or you can also use older version of the plugin

closed 2017-11-23 by me-no-dev