esp8266/arduino-esp8266fs-plugin #43: Arduino.IDE displays "invalid library found"

griffkelleher, 2018-04-30:
my problem is i cant find the library i need and ive tried going to manage library in the Arduino code app but cant find it. PLEASE HELP.

per1234, 2018-04-30:
This is not a library. It's a plugin. If you try to install it as a library then the Arduino IDE will correctly display the "Invalid library" warning. You need to remove the attempted installation in your libraries folder and then carefully follow the installation instructions:
https://github.com/esp8266/arduino-esp8266fs-plugin#installation

zaphodikus, 2020-01-30:
The instructions are wrong, i'm raising a ticket.
<sketchbook directory>/tools/ESP8266FS/tool/esp8266fs.jar
should be same on windows as for ios/unixes, because dropping into
"C:\Users\<username>\Documents\Arduino\tools\ESP8266FS\tool\esp8266fs.jar"
does appear to be good.

Hashan9220, 2021-11-16:
Invalid library found in C:\Users\Hashan\Documents\Arduino\libraries\firebase-arduino-master: no headers files (.h) found in C:\Users\Hashan\Documents\Arduino\libraries\firebase-arduino-master

How to fix this error ?