esp8266/arduino-esp8266fs-plugin #47: Build a plug-in

mcmega, 2018-09-09:
Hello!
Thanks for the plugin!

I need to change some things. I cloned the repository in the eclipse.
How to get the esp8266fs.jar file?

Please make a quick guide to assembling the plugin.

mcmega, 2018-09-09:
I have specified the actual paths, but the jar file is not created.
```
#!/usr/bin/env bash

function pause(){
   read -p "$*"
}

if [[ -z "$INSTALLDIR" ]]; then
    INSTALLDIR="C:/Program Files (x86)/Arduino"
fi
echo "INSTALLDIR: $INSTALLDIR"

lib=$INSTALLDIR/lib

pde_path="C:/Program Files (x86)/Arduino/lib/pde.jar"
core_path="C:/Program Files (x86)/Arduino/lib/arduino-core.jar"
lib_path="C:/Program Files (x86)/Arduino/lib/commons-codec-1.7.jar"
if [[ -z "$core_path" || -z "$pde_path" ]]; then
    echo "Some java libraries have not been built yet (did you run ant build?)"
    return 1
fi
echo "pde_path: $pde_path"
echo "core_path: $core_path"
echo "lib_path: $lib_path"

set -e

mkdir -p bin

pause 'Press [Enter] key to continue...'

javac -target 1.8 -cp "$pde_path:$core_path:$lib_path" -d bin src/ESP8266FS.java

pushd bin
mkdir -p $INSTALLDIR/tools
rm -rf $INSTALLDIR/tools/ESP8266FS
mkdir -p $INSTALLDIR/tools/ESP8266FS/tool
zip -r $INSTALLDIR/tools/ESP8266FS/tool/esp8266fs.jar *
popd

dist=$PWD/dist
rev=$(git describe --tags)
mkdir -p $dist
pushd $INSTALLDIR/tools
zip -r $dist/ESP8266FS-$rev.zip ESP8266FS/
popd
```
The terminal closes very quickly and there are no errors that can be seen.
Pause work to string: javac -target 1.8 -cp "$pde_path:$core_path:$lib_path" -d bin src/ESP8266FS.java
After that, pauses do not help either.
HELP!

marklinoz, 2019-08-05:
how do I create the .jar file, using windows, what compiler do I use