esp8266/arduino-esp8266fs-plugin #54: lastest build (0.4) jar needed

jj11hh, 2019-02-28:
the zip on the release page only contains source files, I tried to compile it but failed because I don't know how. Could you compile it into a single jar, thanks.  :)

danbates2, 2019-03-04:
+1
I'm getting these errors on compile.
```
$ ./make.sh
INSTALLDIR: /Users/admin/Documents/Arduino
find: ../../..//admin/Library/Saved Application State/com.bitrock.appinstaller.uninstall.savedState: Permission denied
find: ../../..//admin/Library/Saved Application State/com.bitrock.appinstaller.uninstall.savedState: Permission denied
find: ../../..//admin/Library/Saved Application State/com.bitrock.appinstaller.uninstall.savedState: Permission denied
Some java libraries have not been built yet (did you run ant build?)
./make.sh: line 13: return: can only `return' from a function or sourced script
pde_path: 
core_path: 
lib_path: 
javac: target release 1.8 conflicts with default source release 1.10
```

earlephilhower, 2019-03-06:
I had to add `-source 1.8` to the existing `-target 1.8` command line option in the makefile to build anything under Ubuntu 18.04 in `make.sh`

You will also need to modify the `find` commands to point to your Arduino IDE installation directory (so it can find the Arduino JARs.

I'm attaching my prebuilt version below, but we should still request that @igrr or @me-no-dev build an "official" version and put it on the release dir.
[ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/2936951/ESP8266FS.zip)

danbates2, 2019-03-06:
Thanks.
I'm interested to see if the uploader is now as esptool.py ;)

julynessi, 2019-03-10:
> I had to add `-source 1.8` to the existing `-target 1.8` command line option in the makefile to build anything under Ubuntu 18.04 in `make.sh`
>
> You will also need to modify the `find` commands to [...]

It's works 👍 .. I edited make.sh as below
------------------------------------
#!/usr/bin/env bash

if [[ -z "$INSTALLDIR" ]]; then
    INSTALLDIR="$HOME/Documents/Arduino"
fi
echo "INSTALLDIR: $INSTALLDIR"

pde_path=/Applications/Arduino.app/Contents/Java/pde.jar
core_path=/Applications/Arduino.app/Contents/Java/arduino-core.jar
lib_path=/Applications/Arduino.app/Contents/Java/commons-codec-1.7.jar
if [[ -z "$core_path" || -z "$pde_path" ]]; then
    echo "Some java libraries have not been built yet (did you run ant build?)"
    return 1
fi
echo "pde_path: $pde_path"
echo "core_path: $core_path"
echo "lib_path: $lib_path"

set -e

mkdir -p bin
javac -target 1.8 -source 1.8 -cp "$pde_path:$core_path:$lib_path" \
      -d bin src/ESP8266FS.java

pushd bin
mkdir -p $INSTALLDIR/tools
rm -rf $INSTALLDIR/tools/ESP8266FS
mkdir -p $INSTALLDIR/tools/ESP8266FS/tool
zip -r $INSTALLDIR/tools/ESP8266FS/tool/esp8266fs.jar *
popd
------------------------------------
then
sudo ./make.sh

Jeroen88, 2019-03-30:
@earlephilhower I downloaded your pre compiled library and replaced my existing ESP8266FS path in /tools with yours. Still I get the error SPIFFS Error: esptool not found!
I have no clue what to do, can you help?

earlephilhower, 2019-03-30:
I'm going to be traveling for a week, so I'm the meantime I suggest making sure you have GIT version and done submodule init/update in the Arduino tree.

>

Jeroen88, 2019-03-30:
@earlephilhower I did a full fresh installation of the hardware/esp8266com directory, inclusing the submodule update and a python get.py. Despite of this the error is there. I did not update the Arduino SDK, I am using 1.8.7. Any ideas?

earlephilhower, 2019-04-11:
@me-no-dev just uploaded a compiled version to the release, so I think this can be closed.