esp8266/arduino-esp8266fs-plugin #61: SPIFFS Create Failed - GLIBCXX_3.4.21 not found

PLSousa, 2019-11-10:
Error "SPIFFS Create Failed!" Someone can help me ?

Error messages:
Arduino: 1.8.10 (Linux), Board: "LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

[SPIFFS] data    : /home/paulo/Arduino/libraries/TFT_eSPI/examples/Smooth Fonts/Font_Demo_3/data
[SPIFFS] size    : 2024
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/home/paulo/arduino-1.8.10/hardware/esp8266com/esp8266/tools/mkspiffs/mkspiffs: /usr/lib/i386-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/paulo/arduino-1.8.10/hardware/esp8266com/esp8266/tools/mkspiffs/mkspiffs)

seteq, 2019-11-25:
I had the same issue on CentOS7 - the binary has been compiled for a newer version of libstdc++ which is not available in some distros.
After compiling mkspiffs by myself and replacing the binary it's working fine.
Give that a try!

PLSousa, 2019-11-30:
Thank you very much. As you suggested, after compiling mkspiffs by myself
and replacing the binary, the arduino-esp8266fs-plugin is working fine.