esp8266/arduino-esp8266fs-plugin #64: SPIFFS Upload failed!

roncoa, 2019-11-15:
After updating the core of esp8266 2.6.1 the tool does not work anymore.
[image]

ArcadeNut, 2019-11-16:
i have to same issue

tihoangyeudau, 2019-11-16:
i have same issue

i486you, 2019-11-16:
i have the same issue

TWJubb, 2019-11-16:
Same issue for me... Reverted to ESP8266 Community version 2.6.0 (from 2.6.1) and the SPIFFS upload worked fine.

tihoangyeudau, 2019-11-17:
> Same issue for me... Reverted to ESP8266 Community version 2.6.0 (from 2.6.1) and the SPIFFS upload worked fine.

but we wish it will work with 2.6.1

SunnyPage, 2019-11-21:
i have same issue

DiegoMisa, 2019-11-21:
same issue

DiegoMisa, 2019-11-21:
downgrade to 2.5.2 and work fine

i486you, 2019-11-21:
I found it out how to upload files by using command line.
(Don't forget to close Serial Monitor window when uploading.
 And my laptop is a macbook. You could find a clue from my solution if you use windows OS.)

1. Pack the folder data that files is located in to upload.
**> ./mkspiffs -c /Users/<username>/Documents/Arduino/Test_SPIFFS/data -b 4096 -p 256 spiffs.bin**
   + Refer to: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spiffs.html
   + Path of mkspiffs: /Users/<username>/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506
 
2. Upload the image file packed.
**> esptool.py --chip esp8266 --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash 0xEB000 spiffs.bin**

     + You might get the error 'A fatal error occurred: Timed out waiting for packet header' after uploading. However, just ignore it if the console says 'Wrote ... ' before the error line, and then check if the file is uploaded by following the next step.
    + 0xEB000 is the start address of SPIFFS. You can find it when you try to upload files by clicking on menu button 'Tools > ESP8266 Sketch Data Upload', and then check '[SPIFFS] address: 0xEB000' on the console like the following captured image. 
[image]

3. Checked if the file uploaded successfully.
**> esptool.py --chip esp8266 --port /dev/cu.SLAB_USBtoUART --baud 115200 read_flash 0xEB000 0x1000 backout.txt**
    + 0x1000 is the size of memory to read.
[image]

I hope that it is helpful for you.

hznupeter, 2019-11-23:
i have same issue in esp8266 2.6.1

RudyFiero, 2019-11-23:
Will this be fixed soon?

smartswitchsio, 2019-11-24:
i have same issue in esp8266 2.6.1

2color, 2019-11-24:
This is probably related to https://github.com/esp8266/Arduino/pull/6765 which was introduced in `2.6.1`

me-no-dev, 2019-11-25:
plugin has been updated to 0.5.0 :) please give it a try

tihoangyeudau, 2019-11-25:
> plugin has been updated to 0.5.0 :) please give it a try

it makes my arduino can not startup

arthurhoch, 2019-11-25:
Arduino 1.8.8, arduino-esp8266fs-plugin 0.40 and esp8266 by ESP8266 Community 2.6.0 works.

RudyFiero, 2019-11-25:
> it makes my arduino can not startup

I can confirm this. It worked before the plugin update. I have both a a Windows installer version and a Portable install. Both will not run.

Arduino 1.8.10 and ESP8266 2.6.1

hznupeter, 2019-11-26:
i've go back to esp8266 2.5.2,it's more safe

me-no-dev, 2019-11-26:
the zip has been updated with earlier java. Please try again :)

RudyFiero, 2019-11-26:
Arduino starts, but still "**SPIFFS upload failed!**"

DiRH, 2019-11-26:
Same issue with 2.5.2.

tihoangyeudau, 2019-11-27:
> Arduino starts, but still "**SPIFFS upload failed!**"

#68

masoud935, 2019-12-01:
hi all... I have a same issue...but I find solve it.
IDE is 1.8.10  and esp is2.5.2
change your jar file to 
[ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)

jpasqua, 2019-12-04:
Also failing for me on esp 2.6.2, ide 1.8.10, and plugin v5 (actually I tried every released plugin and they all fail with different messages). I am not in a position to downgrade to 2.5.2. I will try the manual uploading steps mentioned above to see if I can get that to work until this is fixed.

RudyFiero, 2019-12-04:
To get around the failure I had used the web upload code from the FSbrowser.ino example and put that into the program I wanted to use. I wanted to try out someone else's examples and their program made use of SPIFFS.

jpasqua, 2019-12-04:
Thanks to @i486you for the guidance on the manual process. I have it working now (though I had to download a separate copy of mkspiffs for some reason). I tried to just debug the plugin, but I couldn't find a java version that was compatible with the IDE. It would fail on launch saying my plugin was compiled with too new a version of Java. I backed off versions a bit, but no joy. Hopefully it will be an easy fix. It seems to just be generating an incorrect command line.

coolpixel, 2019-12-05:
> hi all... I have a same issue...but I find solve it.
> IDE is 1.8.10 and esp is2.5.2
> change your jar file to
> [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)

it works for me ... thanks

PersonalityFox, 2019-12-20:
> hi all... I have a same issue...but I found the solution
> IDE is 1.8.10 and esp is2.5.2
> change your jar file to
> [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)

its work with ide 1.8.9
Thank you

Bob0505, 2019-12-23:
> > hi all... I have a same issue...but I found the solution
> > IDE is 1.8.10 and esp is2.5.2
> > change your jar file to
> > [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)
> 
> its work with ide 1.8.9
> Thank you

may I know which version about ESP8266FS.zip ?

PersonalityFox, 2019-12-23:
> > > hi all... I have a same issue...but I found the solution
> > > IDE is 1.8.10 and esp is2.5.2
> > > change your jar file to
> > > [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)
> > 
> > 
> > its work with ide 1.8.9
> > Thank you
> 
> may I know which version about ESP8266FS.zip ?

i dont know
may be @masoud935 knows answer to this question

anctsys, 2019-12-27:
Hi
Just a little precision, for ide arduino 1.8.10 you must download the last ESP8266FS last release on
https://github.com/esp8266/arduino-esp8266fs-plugin/releases/
Bruno

masoud935, 2019-12-27:
> > hi all... I have a same issue...but I find solve it.
> > IDE is 1.8.10 and esp is2.5.2
> > change your jar file to
> > [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)
> 
> it works for me ... thanks

😁😁😁

masoud935, 2019-12-27:
> > > hi all... I have a same issue...but I found the solution
> > > IDE is 1.8.10 and esp is2.5.2
> > > change your jar file to
> > > [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)
> > 
> > 
> > its work with ide 1.8.9
> > Thank you
> 
> may I know which version about ESP8266FS.zip ?

> > > hi all... I have a same issue...but I found the solution
> > > IDE is 1.8.10 and esp is2.5.2
> > > change your jar file to
> > > [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)
> > 
> > 
> > its work with ide 1.8.9
> > Thank you
> 
> may I know which version about ESP8266FS.zip ?

I don't know which version this is.
I just know this version is working properly

anctsys, 2019-12-27:
@masoud935 
> I don't know which version this is.
> I just know this version is working properly

My configuration
IDE is 1.8.10
ESP8266 WEMOS D1 mini 
Flash size: 4MB
esptool.py v2.8

This is my process to upload data into FS for ESP32 and ESP8266
- Add the last library of esp8266fs into tools directory  (https://github.com/esp8266/arduino-esp8266fs-plugin/issues/64)
- Select Erase Flash to Only Sketch to tools for your board configuration. Because erase your data FS after each upload of your sketch
- Upload your data directroy with the tools ESP8266 or 32

roncoa, 2019-12-27:
Solved with
https://github.com/esp8266/arduino-esp8266fs-plugin/releases/download/0.5.0/ESP8266FS-0.5.0.zip

closed 2019-12-27 by roncoa

Bob0505, 2019-12-28:
> hi all... I have a same issue...but I found the solution
> IDE is 1.8.10 and esp is2.5.2
> change your jar file to
> [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)

Dear @masoud935 ,

May I know which version of attach file(ESP8266FS.zip)?
It's work with Arduino IDE Ver.1.8.10

Thank you very much!

rezaajdarkosh, 2019-12-29:
> Hi
> Just a little precision, for ide arduino 1.8.10 you must download the last ESP8266FS last release on
> https://github.com/esp8266/arduino-esp8266fs-plugin/releases/
> Bruno

rezaajdarkosh, 2019-12-29:
Thank you very much!

tonci14, 2020-03-26:
No needed to downgrade, just updated from 
https://github.com/esp8266/arduino-esp8266fs-plugin/releases/

I am using IDE 1.8.12

Aaqu, 2020-04-19:
> No needed to downgrade, just updated from
> https://github.com/esp8266/arduino-esp8266fs-plugin/releases/
> 
> I am using IDE 1.8.12

I using IDE 1.8.12

`sptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'write_flash'`

help

Aaqu, 2020-04-19:
> > No needed to downgrade, just updated from
> > https://github.com/esp8266/arduino-esp8266fs-plugin/releases/
> > I am using IDE 1.8.12
> 
> I using IDE 1.8.12
> 
> `sptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'write_flash'`
> 
> help

Thank you, I need update esp libray to 2.6.3

alireza-azar, 2020-06-24:
> 
> 
> hi all... I have a same issue...but I found the solution
> IDE is 1.8.10 and esp is2.5.2
> change your jar file to
> [ESP8266FS.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/3908067/ESP8266FS.zip)

thank you so much bro this is work for me

omarnas, 2020-09-05:
I have the same issue and my solution was going back to an older version of upload.py (attached below )
[upload.py.zip](https://github.com/esp8266/arduino-esp8266fs-plugin/files/5178418/upload.py.zip) replace it with the existing one (/Users/YOUUSER/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/upload.py)

denimjeans, 2021-12-26:
Version 0.5.0 does not work for me with 18.10 and on Windows 10. 
Filesystem will be created ok, but uploading does not work. 
Workaround as suggested by @i486you helped me. Here is what needed to do:

change to folder:
`C:\Users\YOUR_USER_NAME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\tools\esptool>`
`
esptool.py --chip esp8266 --port com4 --baud 115200 write_flash 0x200000  C:\Users\YOUR_USER_NAME\AppData\Local\Temp\arduino_build_225904\WIFI_IR_BLASTER.spiffs.bin
`
address might be different for your usecase.  You will find all attributes if you scroll little up. 

Hope they will fix it soon, it's kind of messy, but works for me.

denimjeans, 2021-12-28:
Solved: 
I needed to update my ESP development enviroment: 
**Tools-> Board->Boardmanager**, Update ESP8266 to 3.0.2, then try again. It works for me. I was at 2.6.1 before... 
tested on windows 10, Arduino 18.10 and 18.19, ESP8266 v 3.0.2.