esp8266/arduino-esp8266fs-plugin #67: ESP8266FS-0.5.0 makes arduino not startup

tihoangyeudau, 2019-11-25:
when i update to ESP8266FS-0.5.0, it makes arduino not startup, when i delete it , i can open arduino.

staffa007, 2019-11-25:
same problem!

me-no-dev, 2019-11-25:
which version IDE?

arthurhoch, 2019-11-25:
> which version IDE?

1.8.10

staffa007, 2019-11-25:
1.8.10 both Mac OS and Windows

arthurhoch, 2019-11-25:
Arduino 1.8.8, arduino-esp8266fs-plugin 0.40 and esp8266 by ESP8266 Community 2.6.0 works.
For emergency.

baoshuai9527, 2019-11-26:
same problem!windows 7,arduino ide 1.8.9,

me-no-dev, 2019-11-26:
the zip has been updated with earlier java. Please try again :)

tihoangyeudau, 2019-11-26:
it works. i can open arduino. but i cant upload fs, there is the issue:
`[SPIFFS] data    : D:\DESKTOP BACKUP 2\esp8266-fastled-webserver-3-add-artnet\data
[SPIFFS] size    : 1000
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/css/bootstrap.min.css

/css/minicolors.min.css

/css/simple.css

/css/styles.css

/favicon.ico

/fonts/glyphicons.eot

/fonts/glyphicons.svg

/fonts/glyphicons.ttf

/fonts/glyphicons.woff

/fonts/glyphicons.woff2

/images/atom196.png

/index.htm

/js/app.js

/js/bootstrap.min.js

/js/jquery-3.1.1.min.js

/js/minicolors.min.js

/js/r-websocket.min.js

/js/simple.js

/simple.htm

[SPIFFS] upload  : C:\Users\tuant\AppData\Local\Temp\arduino_build_426210/esp8266-fastled-webserver-3-add-artnet.spiffs.bin
[SPIFFS] address  : 0x300000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : COM4
[SPIFFS] speed    : 115200
[SPIFFS] python   : C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
[SPIFFS] uploader : C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\tools\upload.py

usage: esptool write_flash [-h] [--erase-all]

                           [--flash_freq {keep,40m,26m,20m,80m}]

                           [--flash_mode {keep,qio,qout,dio,dout}]

                           [--flash_size FLASH_SIZE]

                           [--spi-connection SPI_CONNECTION] [--no-progress]

                           [--verify] [--encrypt]

                           [--ignore-flash-encryption-efuse-setting]

                           [--compress | --no-compress]

                           <address> <filename> [<address> <filename> ...]

esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'write_flash'

SPIFFS Upload failed!`

tihoangyeudau, 2019-11-26:
> it works. i can open arduino. but i cant upload fs, there is the issue:
> 
> ...
> [SPIFFS] upload : C:\Users\tuant\AppData\Local\Temp\arduino_build_426210/demo.spiffs.bin
> [SPIFFS] address : 0x300000
> [SPIFFS] reset : --before default_reset --after hard_reset
> [SPIFFS] port : COM4
> [SPIFFS] speed : 115200
> [SPIFFS] python : C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
> [SPIFFS] uploader : C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\tools\upload.py
> 
> usage: esptool write_flash [-h] [--erase-all]
> 
> ```
>                        [--flash_freq {keep,40m,26m,20m,80m}]
> 
>                        [--flash_mode {keep,qio,qout,dio,dout}]
> 
>                        [--flash_size FLASH_SIZE]
> 
>                        [--spi-connection SPI_CONNECTION] [--no-progress]
> 
>                        [--verify] [--encrypt]
> 
>                        [--ignore-flash-encryption-efuse-setting]
> 
>                        [--compress | --no-compress]
> 
>                        <address> <filename> [<address> <filename> ...]
> ```
> 
> esptool write_flash: error: argument
> 
> : [Errno 2] No such file or directory: 'write_flash'
> SPIFFS Upload failed!

closed 2019-11-26 by tihoangyeudau