esp8266/arduino-esp8266fs-plugin #68: > with ver 0.5.0 i can open arduino. but i cant upload fs, there is the issue

tihoangyeudau, 2019-11-26:
> it works. i can open arduino. but i cant upload fs, there is the issue:
> 
> ...
> [SPIFFS] upload : C:\Users\tuant\AppData\Local\Temp\arduino_build_426210/demo.spiffs.bin
> [SPIFFS] address : 0x300000
> [SPIFFS] reset : --before default_reset --after hard_reset
> [SPIFFS] port : COM4
> [SPIFFS] speed : 115200
> [SPIFFS] python : C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
> [SPIFFS] uploader : C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\tools\upload.py
> 
> usage: esptool write_flash [-h] [--erase-all]
> 
> ```
>                        [--flash_freq {keep,40m,26m,20m,80m}]
> 
>                        [--flash_mode {keep,qio,qout,dio,dout}]
> 
>                        [--flash_size FLASH_SIZE]
> 
>                        [--spi-connection SPI_CONNECTION] [--no-progress]
> 
>                        [--verify] [--encrypt]
> 
>                        [--ignore-flash-encryption-efuse-setting]
> 
>                        [--compress | --no-compress]
> 
>                        <address> <filename> [<address> <filename> ...]
> ```
> 
> esptool write_flash: error: argument
> 
> : [Errno 2] No such file or directory: 'write_flash'
> SPIFFS Upload failed!

_Originally posted by @tihoangyeudau in https://github.com/esp8266/arduino-esp8266fs-plugin/issues/67#issuecomment-558693093_

coenraadhuman, 2019-11-27:
Experiencing the same situation on Arch Linux.

coenraadhuman, 2019-11-28:
@tihoangyeudau I had a quite look this morning and found that my board manager had an update:

[image]
 
This board version works with the upload utility, just used it.

tihoangyeudau, 2019-11-28:
Thank you. 2.6.2 is very new release so i will try it

closed 2019-11-28 by tihoangyeudau