esp8266/arduino-esp8266fs-plugin #79: No files uploaded on the D1 mini pro

zoidby, 2020-07-09:
Hello

I have said MCU but the data upload does not work. 

If i click »ESP8266 Sketch Data Upload«, i get this output:

```
[SPIFFS] data    : /home/user/ESP_Chart_Web_Server/ESP_Chart_Web_Server/data
[SPIFFS] size    : 15336
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/index.html
[SPIFFS] upload  : /tmp/arduino_build_794715/ESP_Chart_Web_Server.spiffs.bin
[SPIFFS] address  : 0x100000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : /dev/ttyUSB0
[SPIFFS] speed    : 460800
[SPIFFS] python   : /home/user/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3
[SPIFFS] uploader : /home/user/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/upload.py
```
And the status bat shows »SPIFFS Image Uploaded«.

This takes less than a second and after that there are no files on the SPIFFS.

zoidby, 2020-07-10:
Ok i got it working. I deleted all libraries, configs, ... and reinstalled everything. After that, it’s working!

closed 2020-07-10 by zoidby

Warlord3, 2021-02-12:
```
[SPIFFS] data    : D:\Projekte\Bartender2\Bartender\data
[SPIFFS] size    : 2024
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/success.html
/upload.html
/WebSocket.js
[SPIFFS] upload  : C:\Users\X\AppData\Local\Temp\arduino_build_107761/Bartender.spiffs.bin
[SPIFFS] address  : 0x200000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : COM5
[SPIFFS] speed    : 921600
[SPIFFS] python   : C:\Users\X\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
[SPIFFS] uploader : C:\Users\X\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\upload.py

esptool.py v2.8
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: e0:98:06:86:35:fa
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 2072576 bytes to 4558...
Wrote 2072576 bytes (4558 compressed) at 0x00200000 in 0.1 seconds (effective 147322.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

I still have the same problem, even after reinstalling everything. It says upload complete but no file is uploaded

zoidby, 2021-02-12:
@Warlord3 

I can’t tell you how to fix your problem, but SPIFFS is deprecated anyway.

Use [LittleFS](https://github.com/littlefs-project/littlefs) and the [LittleFS-Uploader](https://github.com/earlephilhower/arduino-esp8266littlefs-plugin). Tat works much better. SPIFFS always produced corrupt data when writing and reading from the same file at the same time.

Warlord3, 2021-02-12:
```
[LittleFS] data    : D:\Projekte\Bartender2\Bartender\data
[LittleFS] size    : 2024
[LittleFS] page    : 256
[LittleFS] block   : 8192
/success.html
/upload.html
/WebSocket.js
[LittleFS] upload  : C:\Users\X\AppData\Local\Temp\arduino_build_41615/Bartender.mklittlefs.bin
[LittleFS] address : 0x200000
[LittleFS] reset   : --before default_reset --after hard_reset
[LittleFS] port    : COM5
[LittleFS] speed   : 921600
[LittleFS] python   : C:\Users\X\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe
[LittleFS] uploader : C:\Users\X\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\tools\upload.py

esptool.py v2.8
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: e0:98:06:86:35:fa
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 2072576 bytes to 3778...
Wrote 2072576 bytes (3778 compressed) at 0x00200000 in 0.1 seconds (effective 174532.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
Same result with littleFS.