esp8266/arduino-esp8266fs-plugin #87: SPIFFS Error: esptool not found!

alve89, 2021-04-05:
I'm on macOS 11.0.1. My Arduino IDE is installed in `/Applications`, my Arduino sketchbook path is `/Volumes/Daten/myuser/Documents/Programmierung/Arduino`. I added the ESP8266 support via board manager.

After unpacking the ZIP from [https://github.com/esp8266/arduino-esp8266fs-plugin/releases/tag/0.3.0](https://github.com/esp8266/arduino-esp8266fs-plugin/releases/tag/0.3.0) to `/Volumes/Daten/myuser/Documents/Programmierung/Arduino/tools/` and restarting the IDE the menu item `ESP8266 Sketch Data Upload` was displayed. Clicking it gives the following error:

> SPIFFS Error: esptool not found!

I then tried two different locations to unpack the ZIP to:

1. /Volumes/Daten/myuser/Documents/Arduino (as mentioned in the README
2. /Volumes/Daten/stefanherzog/Library/Arduino15/packages/esp8266 because I thought the ESPTool should be placed somewhere there

In the end nothing worked for me. So: How do I install this plugin properly?

alve89, 2021-04-21:
Really no idea?

Mekacher-Anis, 2021-11-16:
updating to the latest version helped me.