esp8266/arduino-esp8266fs-plugin #90: SPIFFS Upload Failed

jahnavi26052003, 2021-07-07:
Arduino: 1.8.15 (Windows Store 1.8.49.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Multiple libraries were found for "ConfigManager.h"

In file included from C:\Users\JANU VEMPATI\OneDrive\Documents\Arduino\ESP_check\ESP_check.ino:2:

 Used: C:\Users\JANU VEMPATI\OneDrive\Documents\Arduino\libraries\ConfigManager

C:\Users\JANU VEMPATI\OneDrive\Documents\Arduino\libraries\ConfigManager\src/ConfigManager.h:20:10: fatal error: ArduinoJson.h: No such file or directory

 Not used: C:\Users\JANU VEMPATI\OneDrive\Documents\Arduino\libraries\ConfigManager-master

   20 | #include "ArduinoJson.h"

      |          ^~~~~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

[SPIFFS] data    : C:\Users\JANU VEMPATI\OneDrive\Documents\Arduino\ESP_check\data

[SPIFFS] size    : 2024

[SPIFFS] page    : 256

[SPIFFS] block   : 8192

/code_index.html
[SPIFFS] upload  : C:\Users\JANUVE~1\AppData\Local\Temp\arduino_build_576721/ESP_check.spiffs.bin

[SPIFFS] address  : 0x200000

[SPIFFS] reset    : --before default_reset --after hard_reset

[SPIFFS] port     : COM3

[SPIFFS] speed    : 115200

[SPIFFS] python   : C:\Users\JANU VEMPATI\OneDrive\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1\python3.exe

[SPIFFS] uploader : C:\Users\JANU VEMPATI\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.1\tools\upload.py

esptool.py v3.0
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
  File "C:\Users\JANU VEMPATI\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.1\tools\upload.py", line 66, in <module>
    esptool.main(cmdline)
  File "C:/Users/JANU VEMPATI/OneDrive/Documents/ArduinoData/packages/esp8266/hardware/esp8266/3.0.1/tools/esptool\esptool.py", line 3552, in main
    esp.connect(args.before, args.connect_attempts)
  File "C:/Users/JANU VEMPATI/OneDrive/Documents/ArduinoData/packages/esp8266/hardware/esp8266/3.0.1/tools/esptool\esptool.py", line 529, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0x08)
SPIFFS Upload failed!

Can someone help me with this error.

rohitssj, 2023-03-28:
did u got any solution?

AlienHD888, 2023-08-30:
Hi All

Can anibady help me please with Node MCU v3  ???

I do not understand why the connection does not occur if the LED light blinks
Driver CH340G
baud rate 5600
port COM 5

esptool.py v3.0
Serial port COM5
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\Alien\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0/tools/upload.py", line 66, in <module>
    esptool.main(cmdline)
  File "C:/Users/Alien/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/tools/esptool\esptool.py", line 3552, in main
    esp.connect(args.before, args.connect_attempts)
  File "C:/Users/Alien/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/tools/esptool\esptool.py", line 529, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0x08)
esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0x08)