esp8266/arduino-esp8266fs-plugin #96: the esp8266fs file isn't showing up in the tools option

rohitssj, 2023-03-28:


lesleylemmens, 2023-04-02:
same problem here. maybe a to new ide?

saiaravind19, 2023-04-12:
Same here 
I found out that latest release V 0.5.0 is compatible with Arduino IDE version 1.8.5 any later versions the plugin is not working.
Try using the versions mentioned above

FlorianHeigl, 2023-10-02:
Sam  here, definitely compatibility issue with the new IDE version. installing the older one makes no sense, rather gonna have to move to another SPI driver. currently our wedding pics aren't showing after I just wanted to update the Wifi PSK.
so this is sadz. I hope I can get SPIFFS access without wiping

FlorianHeigl, 2023-10-02:
this is the upstream issue: https://github.com/arduino/arduino-ide/issues/58

tl;dr
- plugin framework has completely changed. vscode integration improves, java plugins are not grandfathered in.
- all old gui integrated tools aren't working, but expressif will help fix it
- spiffs is deprecated, too, one ought to use one of the two newer libraries (also no upload menu there as far as i understand)

earlephilhower, 2023-10-05:
Actually, for LittleFS uploads this is the proper IDE 2.x link:  https://github.com/earlephilhower/arduino-littlefs-upload .  It's compatible with the Raspberry Pi Pico and the ESP8266.

I did not make a SPIFFS uploader for 2.x, since SPIFFS is deprecated, but you're free to fork the [littlefs IDE 2.x  plug in](https://github.com/earlephilhower/arduino-littlefs-upload) and do the needful (very straightforward).