esp8266/arduino-esp8266fs-plugin #99: The plugin does not show up in a menu

LSergM, 2023-10-19:
Setup:
Windows 10 Enterprise, version: 22H2
IDE: 2.2.1

It does work with IDE 1.8.19 with the same settings. (Look at the picture below)

[image]

[image]

Paschroeder, 2023-11-18:
Same for me with following Win Version and Arduino IDE 2.2.1:

Edition	Windows 11 Pro
Version	23H2
Installed on	‎07/‎05/‎2022
OS build	22631.2715
Experience	Windows Feature Experience Pack 1000.22677.1000.0

candideu, 2023-11-27:
Ditto

hobus1domini, 2024-01-04:
Hey guys!

Had the same problem. This is not part of this plugin, it's a general Arduino IDE 2 thing. Look here:
https://github.com/arduino/arduino-ide/issues/58

TL;DR: If u need tools / plugins, you are bound to the classic Arduino IDE for now!

candideu, 2024-01-05:
> Hey guys!
> 
> Had the same problem. This is not part of this plugin, it's a general Arduino IDE 2 thing. Look here: [arduino/arduino-ide#58](https://github.com/arduino/arduino-ide/issues/58)
> 
> TL;DR: If u need tools / plugins, you are bound to the classic Arduino IDE for now! One would need a new plugin.

I'm not sure if this will work for everyone's project, but I found a solution for mine using another plugin, and version 2.x of the IDE. Here's my breakdown: https://github.com/iffybooks/pocket-wifi-portal/issues/2#issuecomment-1837755761